flipperdevices/flipperzero-firmware #4148: [Bug] BadUSB input is running too fast

ase1590, 2025-03-12:
### Describe the bug.

The current BadUSB STRING input speed the the Flipper provides in 1.2.0 by default exceed that which Windows can reliably keep up with. 

Running the Demo_Windows script will result in multiple missed characters, as will any script that has long amounts of string input.  

### Reproduction

1. run BadUSB
2. run the demo_windows.txt script 
3. note that the input is too fast, and characters are missed. 

### Target

1.2.0

### Logs

```Text

```

HeavenlyFlight, 2025-03-13:
Noticed the same. For "Install_qFlipper_windows.txt" had to change a DELAY to 3000 and worked like a charm, being the beginning of the file like this 
```
REM Written by @dexv
DELAY 2000
GUI r
DELAY 500
STRING powershell
ENTER
**DELAY 3000**
STRING $url = "https://update.flipperzero.one/qFlipper/release/windows-amd64/portable"
```

However for demo_windows.txt everything goes too fast and I think is more a generalized issue rather than a single lonely DELAY miss time

ase1590, 2025-03-13:
Update on this:
The problem isn't exactly the flipper itself, as downgrading quite far to 0.78 made no difference.

It seems Windows 11 notepad.exe specifically is now too slow to handle the rate of input the flipper types at. all other text editors work more or less without issue. 

For it to reasonably output on notepad.exe, it seems `DEFAULT_STRING_DELAY 10` must be used, otherwise the current version of notepad.exe (Windows Notepad 11.2412.16.0) is simply just too slow at parsing the text being typed in at high speeds now. 

I think the better solution now is just to add that line to the demo_windows.txt and call it a day, since this seems to be notepad specific.