flipperdevices/flipperzero-firmware #4190: Lock f0 in dummy mode

ryderfigga, 2025-04-15:
### Description of the feature you're suggesting.

Hello, 
basicly I want to lock device in dummy mode and enter pin to go to default mode, so locking flipper -> you goin to dummy mode -> enter pin -> you going to default mode

hedger, 2025-04-16:
Does not sound like a good idea.

closed 2025-04-16 by hedger

ryderfigga, 2025-04-16:
Hello, @hedger, may I know why it's not good idea? I think it would be good to pick apps (or set just the passport) that you can use without pin in dummy mode and then after pin combo unlock whole device. Dummy mode is for me useless if you can exit it with just 2 clicks. 

And it was stated on kickstarter 5yrs ago: [https://www.kickstarter.com/projects/flipper-devices/flipper-zero-tamagochi-for-hackers/posts/2920182](https://www.kickstarter.com/projects/flipper-devices/flipper-zero-tamagochi-for-hackers/posts/2920182)

> Additionally, it will be possible to hide some of the Flipper’s functions behind the secret joystick shortcut. And until this combination is entered, Flipper is just a harmless cute Tamagotchi with a bunch of cool mini-games and nothing else.