flipperdevices/flipperzero-firmware #4194: Technical Specification for Flipper Zero Mifare Classic 1K Enhancement

killerbe, 2025-04-19:
### Describe the enhancement you're suggesting.

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-c

[test_original.nfc.json](https://github.com/user-attachments/files/19822212/test_original.nfc.json)

[hf-mf-E3D143BD-dump.json](https://github.com/user-attachments/files/19822215/hf-mf-E3D143BD-dump.json)

[test_E3D143BD.nfc.json](https://github.com/user-attachments/files/19822232/test_E3D143BD.nfc.json)

[96B8A921.nfc.json](https://github.com/user-attachments/files/19822211/96B8A921.nfc.json)
[2573B480.nfc.json](https://github.com/user-attachments/files/19822210/2573B480.nfc.json)

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=RU style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>1.
General Description<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Project
Name:</span></b><span lang=EN-US style='mso-ansi-language:EN-US'> Enhancing
Flipper Zero functionality for ISO14443-A MIFARE Classic 1K cards<o:p></o:p></span></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Project
Goal:</span></b><span lang=EN-US style='mso-ansi-language:EN-US'> Eliminate
issues identified during testing of Flipper Zero with MIFARE Classic 1K tags,
especially with MF3 filter<o:p></o:p></span></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Initial
Problem:</span></b><span lang=EN-US style='mso-ansi-language:EN-US'> Testing of
ISO14443-A MIFARE Classic 1K standard revealed critical functionality
deficiencies in Flipper Zero that prevent successful copying and emulation of
these tag types<o:p></o:p></span></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>2.
Identified Issues (Current State)<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>2.1
Scenario 1: Incorrect Standard Key Copying<o:p></o:p></span></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l2 level1 lfo1;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Reproduction Conditions:</span></b><span
     lang=EN-US style='mso-ansi-language:EN-US'> Scanning an original finalized
     tag with MF3 filter using standard Key A and Key B (FF FF FF FF FF FF)<o:p></o:p></span></li>
 <li class=MsoNormal style='mso-list:l2 level1 lfo1;tab-stops:list 36.0pt'><b>Observed
     Behavior:</b> <o:p></o:p></li>
 <ul style='margin-top:0cm' type=circle>
  <li class=MsoNormal style='mso-list:l2 level2 lfo1;tab-stops:list 72.0pt'><span
      lang=EN-US style='mso-ansi-language:EN-US'>Flipper Zero fails to
      correctly copy the keys<o:p></o:p></span></li>
[... 119 lines cut ...]
     mechanism with signal parameter variation<o:p></o:p></span></li>
 <li class=MsoNormal style='mso-list:l3 level1 lfo6;tab-stops:list 36.0pt'><span
     lang=EN-US style='mso-ansi-language:EN-US'>Develop a diagnostic and
     logging system for reading issues with per-sector details<o:p></o:p></span></li>
 <li class=MsoNormal style='mso-list:l3 level1 lfo6;tab-stops:list 36.0pt'><span
     lang=EN-US style='mso-ansi-language:EN-US'>Implement visual indication of
     individual sector reading status in the user interface<o:p></o:p></span></li>
</ul>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>4. Test
Cases<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Test
Case 1: Working with Standard Keys<o:p></o:p></span></b></p>

Action | Data | Expected Result
-- | -- | --
Scan   original finalized tag with MF3 filter | Key A and   Key B: FF FF FF FF FF FF | Successful   reading and copying of keys with all sector data
Check UID   in the obtained dump |   | Correct   UID (2573B480) should be read and saved without distortion
Write   data to empty MF3 tag | UID: 2573B480 | Tag   should be correctly written with all sectors and keys
Verify   functionality of written tag |   | Copy   should successfully open intercom door similar to the original

</body>

</html>

killerbe, 2025-04-19:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=RU style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Test
Case 2: Recovery of Unknown Keys<o:p></o:p></span></b></p>

Action | Data | Expected Result
-- | -- | --
Scan   finalized tag with MF3 filter | UID: 96B8A921, unknown keys | Key   recovery process should start with progress indication
Measure key recovery time |   | Keys   should be recovered within a time not exceeding 2 minutes (comparable to   ikeybase express performance)
Check recovery result |   | Correct   keys should be determined (Key A: EF 41 40 37 DC F6, Key B: B6 DA 90 FE 2D   E0)
Read   sector contents using recovered keys |   | All   sectors should be successfully read and saved to dump
Write   data to empty MF3 tag |   | Tag   should be correctly written with all sectors and keys
Verify   functionality of written tag |   | Copy   should successfully open intercom door similar to the original

</body>

</html>

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=RU style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNormal><b>Test Case 3: Sector Reading Stability<o:p></o:p></b></p>

Action | Data | Expected Result
-- | -- | --
Scan   various tag types with MF3 filter | Set of   test tags from different manufacturers | All   sectors of all tag types should be successfully read without errors or   omissions
Display sector reading status |   | Interface   should display status of each sector with indication of problematic areas
Check retry mechanism |   | When   problematic sectors are detected, system should automatically perform retry   attempts with parameter adaptation
Repeat scanning several times |   | Reading   results should be stable and identical between consecutive scans
Check diagnostic log |   | Log   should contain detailed information about reading process with parameters and   status for each sector

</body>

</html>

bettse, 2025-04-22:
That video is of a custom firmware....

killerbe, 2025-04-22:
> That video is of a custom firmware....

Hello.
Yes, I'm aware that the firmware is not official, and I'm writing in the official branch.
But I really don't want to spend a lot of time on reinstallations, since in terms of NFC, the unofficial firmware differs very little regarding standard functions, like reading for example. Regarding any fixes, I've compared the code of both firmwares.

dogtopus, 2025-05-21:
There used to be a hardnested FAP that can do maybe 50% of what you asked for, but it's not completely standalone due to the amount of computation required, and it broke during one of the firmware mass refactoring, to the point that IIRC can't be resolved without a complete rewrite.

If getting your hands on the reader software and reverse-engineering it is impossible, I'd suggest that you switch to Proxmark3 paired with a portable PC (like an x86 gaming handheld), but even then if you encounter those static (encrypted) nonce tags, it would srill take forever to crack them.