flipperdevices/flipperzero-firmware #4196: Technical Specification: NFC Tag Information Application for Flipper Zero

killerbe, 2025-04-19:
### Description of the feature you're suggesting.

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=RU style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>NFC Tag
Info Application for Flipper Zero<o:p></o:p></span></b></p>

<p class=MsoNormal><b>Project Overview<o:p></o:p></b></p>

Parameter | Value
-- | --
Project Name | NFC   Tag Info for Flipper Zero
Project Type | Application   for Flipper Zero device
Priority | High
Status | New

<p class=MsoNormal><b><span style='color:white;mso-color-alt:windowtext'>Objective</span><o:p></o:p></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Develop
an application for Flipper Zero that reads and displays detailed technical
information about NFC tags (similar to Proxmark, TMD-5S, or SMKey), including
filter detection.<o:p></o:p></span></b></p>

<p class=MsoNormal><b>Functional Requirements<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l4 level1 lfo1;tab-stops:list 36.0pt'><b>Core
     Functions:<o:p></o:p></b></li>
 <ul style='margin-top:0cm' type=circle>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Scan ISO14443-A NFC tags and
      other supported standards<o:p></o:p></span></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Display: UID, ATQA, SAK, tag
      type, RATS support status, static nonce info, and user hints<o:p></o:p></span></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Detect filter presence and
      type (MF1, MF2, MF3, OTP, ZERO, etc.)<o:p></o:p></span></b></li>
 </ul>
 <li class=MsoNormal style='mso-list:l4 level1 lfo1;tab-stops:list 36.0pt'><b>UI
     Requirements:<o:p></o:p></b></li>
 <ul style='margin-top:0cm' type=circle>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Structured output similar to Proxmark
      console<o:p></o:p></span></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b>Clear
      data hierarchy with separators<o:p></o:p></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b>Ability
      to save scan results<o:p></o:p></b></li>
 </ul>
</ul>

<p class=MsoNormal><b>Non-functional Requirements<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Fast scanning (&lt;3 seconds
     per tag)<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b>Proper
     error handling<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b>Low
     power consumption<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b>Standard
     Flipper Zero navigation<o:p></o:p></b></li>
</ul>

<p class=MsoNormal><b>Test Cases<o:p></o:p></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Test
Case #1: Scanning MIFARE Classic 1K<o:p></o:p></span></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l2 level1 lfo3;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Action: Launch app, scan MIFARE
     Classic 1K tag<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l2 level1 lfo3;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Data: ISO14443-A MIFARE Classic
     1K tag<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l2 level1 lfo3;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Expected: Structured display of
     UID, ATQA, SAK, tag type, RATS support, static nonce, and tag validity
     status<o:p></o:p></span></b></li>
</ul>

<p class=MsoNormal><b>Test Case #2: Filter Detection<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l0 level1 lfo4;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Action: Launch app, scan tag
     with known filter<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l0 level1 lfo4;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Data: NFC tag with MF1 filter<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l0 level1 lfo4;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Expected: Display &quot;Filter
     detected: MF1&quot; and other tag information<o:p></o:p></span></b></li>
</ul>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Technical
Issues to Resolve<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Based on
previous implementation attempts, these issues need fixing:<o:p></o:p></span></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l3 level1 lfo5;tab-stops:list 36.0pt'><b>FWT
     Timeout errors<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l3 level1 lfo5;tab-stops:list 36.0pt'><b>ATS
     request failures<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l3 level1 lfo5;tab-stops:list 36.0pt'><b>ATS
     reading errors<o:p></o:p></b></li>
</ul>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Previous
implementation attempted to determine filter type but encountered errors in
debug logs:<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>122528
[D][Nfc] FWT Timeout<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>122579
[E][Iso14443_4aPoller] ATS request failed<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>122582
[D][Iso14443_4aPoller] Failed to read ATS<o:p></o:p></span></b></p>

<p class=MsoNormal><b>Technical Recommendations<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Use NFC libraries from Flipper
     Zero SDK<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Work with low-level API for
     precise scan control<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b>Implement
     timeouts and retry logic<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b>Add
     detailed logging<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Study existing implementation
     to understand current limitations<o:p></o:p></span></b></li>
</ul>

<p class=MsoNormal><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

</body>

</html>

bettse, 2025-04-22:
>  This should be a standalone application that quickly identifies the tag type before proceeding to read it with the scanner.

you said you work primarily with MFC.  To read a MFC "quickly" requires the keys to the tag.  Do you know the KDF of the system?  Also, for the non-MFC case, what are you expectations?

killerbe, 2025-04-22:
> you said you work primarily with MFC. To read a MFC "quickly" requires the keys to the tag. Do you know the KDF of the system? Also, for the non-MFC case, what are you expectations?

Hello, dear Betsy.

- I don't need to read with keys, but rather an application that will immediately check information about the card, like in proxmark3 with the command 'hf search' or 'hf 14a info'. Screenshot attached.
- If the card is not NFC at 13.56MHz, but for example 125kHz, it would be great to determine information that the frequency is 125kHz and also output some information, similar to how the NFC/Rfid detector application works, but this works with readers, not cards.

Regarding KDF, I am aware it's Key Derivation Function. By the way, you can calculate the key knowing the UID in Mifare Classic 1k, at least 4 bytes for sure. This was done by the Russian-speaking company ikeybase, but for some reason no one has been able to break this algorithm, and they keep it secret. If you have any hints on how it works, how the prediction key is calculated by UID, I would be very grateful.

In the Momentum project, in previous branches with the developer Willy-JL, we couldn't have a constructive and adequate dialogue. There is also my fault, as English is not my primary language, and I made some mistakes in the logic of words.
For example, I wrote in my post that I needed to copy the original key, this was not complete information, I meant to read it and dump it first.)) And without a key, this cannot be done. I also know about magic gen1,2 cards, what blanks to write to and how.

P.S. And yes, in the previous video where I supposedly couldn't stand it when I was going through the keys and didn't wait for completion, I was too lazy to edit the video. I ran it through Proxmark3 with large dictionaries more than once, it didn't help. In the Flipper Telegram chat, someone helped me, calculated the key through the KDF vulnerability using the UID, and I was able to read and dump the original tag. He had a private API, which he also keeps secret and cannot share for a number of reasons. That's why I want to understand, I have one clue... After all, the Flipper Zero community is large, I hope someday they will reveal the secret of KDF.  💯 :D

[image]
[image]

bettse, 2025-04-22:
>  This was done by the Russian-speaking company ikeybase, but for some reason no one has been able to break this algorithm, and they keep it secret. If you have any hints on how it works, how the prediction key is calculated by UID, I would be very grateful.

Reading will never be fast without this algorithm.