flipperdevices/flipperzero-firmware #4200: Enhance SubGhz with protocol command cycling

Nvidiacerv, 2025-04-21:
### Describe the enhancement you're suggesting.

**1. Goal:** Enhance the SubGHz signal emulation functionality to allow users to transmit different commands associated with a specific device/protocol, beyond simply replaying the exact command that was originally captured and saved.
**2. Current Limitation:** Currently, if a user captures and saves, for example, the "Close" command for a Dooya blind remote, the "Send" function can only re-transmit that exact "Close" command. To send "Open" or "Stop", the user must capture and save those specific signals separately or edit the sub file manually for instance using an editor on a pc.
**3. Desired Enhancement:** When a user selects a saved SubGHz key for emulation (sending), the system should:
    • Identify Supported Commands: Determine if the protocol associated with the saved key supports multiple distinct commands (e.g., On/Off, Open/Close/Stop).
    • Present Options (If Applicable): If multiple commands are supported, present the user with a choice of available commands for that protocol before transmission. This list should use human-readable names. The user should be able to select another command for instance by using the up or down button on the flipper zero while in the emulation screen.
    • Generate Selected Command: Upon user selection of a command (which might be the original command or a different one from the list), the system must generate the correct SubGHz signal payload corresponding to the selected command when send is pressed.
    • Reuse Identification Data: Crucially, the newly generated signal must utilize the essential identification parameters (like Device ID, Remote Address, Channel, Key, Counter - depending on the protocol) from the original saved key. This ensures the command is sent as if from the original remote and targets the correct receiving device.
**4. Examples for the Dooya and Intertechno_V3 protocols. But should be workign with all known commands with multiple commands:**
    • Dooya Protocol (Blinds/Curtains):
        ◦ User saves a key captured from the "Close" button.
        ◦ User selects this key and chooses "Send".
        ◦ The system recognizes it's Dooya and presents options: "Open", "Close", "Stop".
        ◦ User selects "Open".
        ◦ The system generates the "Open" command signal, using the Device ID and other necessary data from the saved "Close" key, and transmits it.
    • Intertechno V3 Protocol (Power Outlets):
        ◦ User saves a key captured from the "On" button for a specific outlet address.
        ◦ User selects this key and chooses "Send".
        ◦ The system recognizes it's Intertechno V3 and presents options: "On", "Off".
        ◦ User selects "Off".
        ◦ The system generates the "Off" command signal, using the address/ID from the saved "On" key, and transmits it.
**5. Benefit:** This feature allows users to fully control devices supporting multiple commands using only a single saved key for that device, significantly improving usability and reducing the need to capture every possible command variation. It transforms the Flipper from a simple repeater into a more versatile remote control for supported protocols.

Nvidiacerv, 2025-04-22:
In addition to the above enhancement, an alternative of the cycling command option is a feature that automatically generates opposite commands for a given protocol after decoding a raw signal and being shown in the UI. 

After decoding a raw signal, the app should not only display the decoded signal with the originally received command in the UI but also automatically generate the opposite command of the same protocol and show the additional response in the UI. These additional signals should use the same Device ID and Channel as the original decoded signal.

Skorpionm, 2025-04-29:
this is a great idea for a separate application. because a lot of protocols on different devices are used not exactly as intended by the developers, but simply as it turned out to add them there. and adding such functionality in sub-ghz for some protocols and not adding it for others is not correct and in terms of navigation, this will greatly complicate it

Nvidiacerv, 2025-04-29:
I agree. Meanwhile, I have played around with some ideas.

1) I tried to modify the current SubGHz app to not only display one decoded signal in the UI, but if a command is detected that has an opposite command, such as up/down, open/close, or on/off, an additional signal is displayed in the UI with that opposite command, preserving information like Sn, Ch, etc.

2) A second option I'm experimenting with is creating a custom app to browse and select a previously saved .sub file. Then, based on the protocol information, display the opposite or available commands, let the user select one, and transmit that command using the same Sn, Channel etc..

Currently i am focussing on the 2nd option and will share it if i get it to work. Would be great to have this feature within this firmware. Saves me, and hopefully others, to capture and process decoded signals into my home automation system.