flipperdevices/flipperzero-firmware #4217: NFC doesn't work at all, RFID is fine

b1naryTR, 2025-05-08:
### Describe the bug.

Both versions on dev-5b911f54 and release-1.3.4 versions NFC both reading and emulation fails. I have tried other firmware versions but to no avail. I'm thinking this has something to do with either radio firmware or hardware. I have attached the logs in the issue.

[flipper-log.txt](https://github.com/user-attachments/files/20102792/flipper-log.txt)

### Reproduction

Literally nothing NFC related works, no reproduction.

### Logs

```Text

```

bettse, 2025-05-11:
What card(s) are you testing it with?  Have you used it/them with the flipper before?

b1naryTR, 2025-05-12:
> What card(s) are you testing it with?  Have you used it/them with the flipper before?  

None of the cards I tried worked, logs doesn't even change. I have tried an Classic 1k, Desfire EV1 and an IBM JCOB, I can read all of them with my phone or with ARC122U with libnfc and they all work.

b1naryTR, 2025-05-19:
This was an hardware issue, I hit the flipper few times and it fixed itself.

closed 2025-05-19 by b1naryTR