flipperdevices/flipperzero-firmware #4218: [Buetooth] Why is the extra beacon size limited to 31?

steam3d, 2025-05-08:
Hi. In the main page description, it says that Flipper uses Bluetooth LE 5.4 (possibly STM32WB55), but I did not find the exact model. Does it really support only 31 bytes of beacon size?

It looks like it uses Legacy Advertising, but can we get Extended Advertising with up to 255 bytes of beacon size?

https://github.com/flipperdevices/flipperzero-firmware/blob/5b911f5405f104ad3e3051aa70e2a50a5b2a6d16/targets/f7/ble_glue/extra_beacon.c#L119

hedger, 2025-05-20:
I think we had issues with larger payload sizes back then. Check out AN5270 and AN5289 for reference.

If you can confirm that increased payload sizes indeed work now, we can incorporate the fix in the firmware.

Just curious, what's your usecase for that?

WillyJL, 2025-06-29:
> It looks like it uses Legacy Advertising, but can we get Extended Advertising with up to 255 bytes of beacon size?

no, that requires either the full or the extended BLE stack. i dont remember the exact details right now but i tried to get it working last year and it just was not included in the light stack.